Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #1: Testing without Batteries

lucky01001, 2019-11-27:
I'm now at page 55 of the guide and it says "the car must be equipped with batteries", but i can't see why. Would it harm the device if i push both power switches now?

ZiJace, 2019-12-07:
The equipment will not be damaged, because the test needs to supply power to the motor,servo and other equipment,so the battery needs to be installed.

ZiJace, 2019-12-07:
@lucky01001 The equipment will not be damaged, because the test needs to supply power to the motor,servo and other equipment,so the battery needs to be installed

lucky01001, 2019-12-21:
hey @ZiJace  thanks for your answer! So can I safely run the tests with the power supply instead of the batteries?

ZiJace, 2019-12-24:
@lucky01001 No, the battery must be installed for testing.The data line power supply can only let raspberry pi work.

DenzelChen, 2020-07-23:
@lustoerk I am very sorry that we did not check the issues on GitHub. It has been so long since your last comment. If there is anything we can do, please feel free to contact us via support@freenove.com

closed 2020-07-23 by DenzelChen