Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #17: wrong screw

lzh626, 2021-04-05:
Hi, questions on the Tutorial.pdf for the assembling. Tutorial.pdf, page 72, step 9. Are you sure that M2.5*8 is the right size for the screw? The holes in the white plastic are too small, impossible to let go through the M2.5*8 screws, M1.4*4 screws are more than enough.

lzh626, 2021-04-06:
page 52,  Install Pan Tilt on smart car board, step 2. To be consistent the installation of servo 1, should use M2*4 included in the package to fix servo 0 as for the servo 1. No need to use M2*10+Nut M2.

lzh626, 2021-04-06:
Page 54, Wiring. To be more practical, Wiring should be done before assembling the Pan Tilt. That is, page 54 should happen before page 51. It's hard, if not impossible to plug the wires when there's almost no space at all for the wiring.

DenzelChen, 2021-04-07:
Dear customer,
Thank you so much for your detailed information.
We will check them more carefully and update the tutorial.
Sincerely,
Freenove