Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #18: How to get the battery out?

lzh626, 2021-04-05:
[image]
The design of the battery box is very convenient. It was hard enough to push the battery in, considering the type of battery your suggested/required and the risk to break the plastic. It's impossible to get the battery out in order to re-charge them. 

Is there any trick to make this easy?

DenzelChen, 2021-04-07:
Dear customer,
You can put a wire or rope in battery holder before install batteries. When you want to remove the batteries just pull the two end of the wire.
[image]

lzh626, 2021-04-10:
> Dear customer,
> You can put a wire or rope in battery holder before install batteries. When you want to remove the batteries just pull the two end of the wire.
> [image]

Better to add this image in the Tutorial as well.

amdroid007, 2021-11-02:
This is my solution and it works pretty well - just wrap an electrical tape around the batteries (I found it works better if the tape is towards the top of the battery so you can grab the tape, pull down so the battery presses on the spring, then pull the battery out). See picture.
[image]

MallocArray, 2021-11-30:
@amdroid007 I was really excited to try this out as it looked like it would be effective, but in our class, the electrical tape caused too much friction to slide past the plastic arms on the battery holder and caused the arms to try to bend in with the battery.  We ended up removing the tape so we could insert them.

We have sewn a small loop of ribbon that we put the batteries through so it is under the batteries and has a rest of the loop available to grab to pull up so we don't have loose ends or risk the ribbon slipping out since it is a full circle.  It isn't perfect, but it works.

In future versions, maybe there can be a battery door of sorts so the arms aren't needed at all.