Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #22: Camera is not working

SarthakJaingit, 2021-11-30:
Hi, 

The raspistill -o image.jpg does not work and also in the Client the camera does not work. But when I run libcamera-hello in the ssh then the camera image is captured and displayed in the VNC. Is there is a reason why this is the case? 

regards, 
Sarthak Jain

DenzelChen, 2021-11-30:
You can use this command to check your system.
_cat /etc/rpi-issue_ 

The new system 10-30 doesn't support raspistill -o any more.
You can download to use our latest tutorial.

DeterminedLife, 2021-12-05:
On 2021-10-30 Raspberry Pi version, 11 updates (Bullseye) added a new default camera subsystem based on _libcamera_.  _libcamera-still_ an_d libcamera-vid_ have relaced _raspistill_ and _raspvid_

DeterminedLife, 2021-12-09:
Tutorial requires information on how to run the Bullseye patch :)

DenzelChen, 2021-12-09:
Please download the lastest tutorail. There are information for this on page 43.

tyjtyj, 2022-01-06:
> Please download the lastest tutorail. There are information for this on page 43.

i found the tutorial on page 80.. here is my output for camera

pi@raspberrypi:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server $ vcgencmd get_camera
supported=1 **detected=1**
pi@raspberrypi:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server $ python camera.py

Traceback (most recent call last):
  File "camera.py", line 4, in <module>
    camera.capture('image.jpg')
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1423, in capture
    'Timed out waiting for capture to end')
picamera.exc.PiCameraRuntimeError: Timed out waiting for capture to end

DenzelChen, 2022-01-06:
Please contact us at support@freenove.com