Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #25: can't run servo.py

gchilds, 2021-12-28:
I can't run servo.py, I get file not found for I'm not sure which file.
Led and buzzer worked, but only after I ran `sudo pip3 install rpi_ws281x`

Which file is it looking for?

Thanks

Output
```
pi@raspberrypi:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server $ sudo python3 servo.py 
Now servos will rotate to 90°.
If they have already been at 90°, nothing will be observed.
Please keep the program running when installing the servos.
After that, you can press ctrl-C to end the program.
Traceback (most recent call last):
  File "servo.py", line 34, in <module>
    pwm=Servo()
  File "servo.py", line 5, in __init__
    self.PwmServo = PCA9685(0x40, debug=True)
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/PCA9685.py", line 29, in __init__
    self.bus = smbus.SMBus(1)
FileNotFoundError: [Errno 2] No such file or directory
```

MallocArray, 2021-12-28:
Did you run both build.sh and 'python setup.py' from the Code folder? That should have installed the ws2812 module for you and possibly others that are needed. Run with sudo to make sure they are in the right user folders.

gchilds, 2021-12-29:
I had not done those things!

To run `build.sh` I had to 
`chmod 700 build.sh`
and get to a clean slate with with 
`sudo rm -rf /root/Freenove_RPI_WS281x_Python /usr/local/lib/python3.7`

setup tools were missing, so I did
`sudo apt-get install python3-setuptools` too. 

Then 
`sudo ./build.sh` and `sudo python setup.py` succeeded.

`sudo python servo.py `

then produced

```
pi@raspberrypi:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server $ sudo python3 servo.py 
Traceback (most recent call last):
  File "servo.py", line 2, in <module>
    from PCA9685 import PCA9685
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/PCA9685.py", line 5, in <module>
    import smbus
ModuleNotFoundError: No module named 'smbus'
```

`sudo apt-get install python3-smbus`

fixed that, which got me back to 

```
pi@raspberrypi:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server $ sudo python servo.py 
Now servos will rotate to 90°.
If they have already been at 90°, nothing will be observed.
Please keep the program running when installing the servos.
After that, you can press ctrl-C to end the program.
Traceback (most recent call last):
  File "servo.py", line 34, in <module>
    pwm=Servo()
  File "servo.py", line 5, in __init__
    self.PwmServo = PCA9685(0x40, debug=True)
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/PCA9685.py", line 29, in __init__
    self.bus = smbus.SMBus(1)
FileNotFoundError: [Errno 2] No such file or directory
```

so I'm back to the same situation

gchilds, 2021-12-29:
I2C wasn't enabled

`sudo raspi-config nonint do_i2c 1`

fixed it. I got this answer from Shaynee at the support mail.

closed 2021-12-29 by gchilds