Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #26: python test.py Motor

tyjtyj, 2022-01-01:
The whole system lost power when turn left/right

I tested with different battery and they are panasonic NCR. 

```
The photoresistor voltage on the left is 1.57V
The photoresistor voltage on the right is 1.6V
The battery voltage is 8.16V
```

```
pi@raspberrypi:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server $ sudo python test.py Motor
Program is starting ...
The car is moving forward
The car is going backwards
The car is turning left
```
Each time turn left it die. Even I remove the 2 connector to the motor

MallocArray, 2022-01-01:
Sometimes the batteries are only just connecting and when the robot turns, it shakes the entire body enough to slightly disconnect it.  Try pushing on your batteries to ensure they are firmly connected to the metal pads, like you are the spring pushing it a bit more forward.

tyjtyj, 2022-01-01:
So i managed to test further by powering pi using powerbank. 

With 2 motor the test always passed. 
(This tested with removing one by one  motor each time to rule out it is not single motor issue. more to controller issue)
with 3 motor the test sometime pass left and stop at right. 
with 4 motor 100% wont pass left. 

The car able to move forward with 4 wheel moving. Reverse is working fine. 
This give me believe there is hardware failure on over on turning. ?