Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #27: Rasberry pi not switching on with batteries

suhasshastry, 2022-01-18:
I accidentally connected the raspberry pi to the power supply when it was mounted onto the car. Now the raspberry pi won't start with the batteries from the car. Did I damage the 4wd car circuit? Power seems to be not flowing into the pi.

PS: Rasberry pi is fine with a direct power connection.

MallocArray, 2022-01-18:
You are using both power switches on the robot board, right?

suhasshastry, 2022-01-18:
Yeah. both the switches. The green light of pi is never turned on.

DenzelChen, 2022-01-18:
Please fully charge your batteries to try again.

suhasshastry, 2022-01-19:
One of the batteries is not withstanding the charge. I will try with new batteries.

suhasshastry, 2022-01-29:
One of the batteries was fried. Working fine with new batteries. Thanks @DenzelChen

closed 2022-01-29 by suhasshastry