Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #28: Images from Freenove 4wd Camera are black.

SarthakJaingit, 2022-01-20:
Hi, 

I had a quick issue when trying to use the camera of the Freenove Rover that I assembled. 
I have connected the camera to my raspberry pi on the rover and I created a custom camera script the code is shown below: 

```

import cv2 
import numpy as np 

vid = cv2.VideoCapture(0)
ret, frame = cap.read()
frame = cv2.cvtColor(frame, cv.COLOR_BGR2RGB)

# If I run the bottom commands I get a saved black image

cv2.imwrite("test.jpeg", frame) 

# If I run this instead, I get a very dark image that is very blurry with half of the image black and with pink lines running through it and a lot of time a dark screen too. 

cv2.imshow("frame", frame)
cv2.waitKey(0)
```

I have peeled off the red tape from the camera. I was wondering if the Camera works well with opencv or if there is an alternative that I can use to create a custom camera script. Also, does seeing black images from the code above indicate a problem with software or hardware. I have tried to reconnect the camera band several times, but no change seems to happen with the results. 

Thanks in advance, 
Sarthak Jain

DenzelChen, 2022-01-20:
Please use this command to check camera:
vcgencmd get_camera

SarthakJaingit, 2022-01-20:
> Please use this command to check camera: vcgencmd get_camera

Hi.

I ran the command and got the following output. 

supported=0 detected=0

DenzelChen, 2022-02-21:
Sorry, would you mind contacting us via support@freenove.com?