Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #30: Server is off ?

Sai9555, 2022-02-03:
[image]

Sai9555, 2022-02-03:
I am trying to connect to the server but all i get is this.

lorenzosalvi92, 2022-02-19:
Hi! I am having a similar problem. I don't know whether the issues we are having are connected
[image]
.

saikesava, 2022-02-21:
don't hit the on the button the server connects automatically, I ran into some trouble with the desktop connection but I would recommend the ios or android app that they made to pair with the device. I personally used an iPhone so I am not totally sure about the android side of life.

DenzelChen, 2022-02-21:
> [image]

Just run the code. Don't click the server. The server is turned on automatically.

DenzelChen, 2022-02-21:
> Hi! I am having a similar problem. I don't know whether the issues we are having are connected [image] .

Just run the code. Don't click the server. The server is turned on automatically.

lorenzosalvi92, 2022-02-21:
Hi everybody and thanks a lot for your replies! The problem is that I did not click the On-button. It's just not working and I can't figure out why.

lorenzosalvi92, 2022-02-21:
> > Hi! I am having a similar problem. I don't know whether the issues we are having are connected [image] .
> 
> Just run the code. Don't click the server. The server is turned on automatically.

Hi everybody and thanks a lot for your replies! The problem is that I did not click the On-button. It's just not working and I can't figure out why.