Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #31: I/O error

TheClergy, 2022-04-01:
I am getting this error when trying to run this command I just set up the pi car and may have missed a step but can't seem to find out what it is 

sudo python servo.py
Now servos will rotate to 90°.
If they have already been at 90°, nothing will be observed.
Please keep the program running when installing the servos.
After that, you can press ctrl-C to end the program.
Traceback (most recent call last):
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/servo.py", line 34, in <module>
    pwm=Servo()
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/servo.py", line 5, in __init__
    self.PwmServo = PCA9685(0x40, debug=True)
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/PCA9685.py", line 32, in __init__
    self.write(self.__MODE1, 0x00)
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/PCA9685.py", line 36, in write
    self.bus.write_byte_data(self.address, reg, value)
OSError: [Errno 121] Remote I/O error

DenzelChen, 2022-04-02:
Please make sure you have turned on switches of the car. And you can try to reconnect the connector.
Or please contact us via support@freenove.com

janrutger, 2022-04-02:
I think an issue on I2C bus level. 
Please check 
>> lsmod | grep i2c

and
>> i2cdetect -y 1

As explained on page 43 of the manual
And make sure you turned the switches on