Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #32: Can't start main.py

iXenios, 2022-04-05:
i wanna start the server on the raspberry pi and connect with my phone
but after i do:
sudo python main.py

there comes:
see screenshot

[image]

iXenios, 2022-04-05:
OSError: [Errno 99] Cannot assign requested adress
thats the last part in english

DenzelChen, 2022-04-06:
Did you connect your Raspberry Pi to WIFI?

ShuboOda, 2023-05-13:
I'm having error too when running main.py 

Can you take a loo please

[image]

Shaynee-MO, 2023-05-17:
[image]
Please run the libraries installation program and try again.
sudo python setup.py