Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #34: Camera issue on 64 bit with bullseye

crypto-titan, 2022-05-19:
pi@raspberrypi:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server $ lsb_release -a 
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

pi@raspberrypi:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server $ sudo python camera.py 
Traceback (most recent call last):
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/camera.py", line 1, in <module>
    from picamera import PiCamera
  File "/usr/local/lib/python3.9/dist-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/usr/local/lib/python3.9/dist-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/usr/local/lib/python3.9/dist-packages/picamera/mmal.py", line 49, in <module>
    _lib = ct.CDLL('libmmal.so')
  File "/usr/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmmal.so: cannot open shared object file: No such file or directory

crypto-titan, 2022-05-19:
I've run the patch but did not seem to fix the issue

DenzelChen, 2022-05-20:
Please contact us via support@freenove.com

Kakcalu13, 2024-04-09:
did this fixed? bullseye 12 stoppped working with legacy camera

(这个问题修复了吗？Bullseye 12 停止与旧版本相机合作。)