Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #38: How to integrate more ultrasonic sensors?

Sharath-Lokesh, 2022-11-17:
Hi,
I am trying to add 2 more ultrasonic sensors.
There is only one trigger pin, echo pin.
Is there a way to add 2 more sensors to the bot?

Thanks in advance for your help.

Sharath-Lokesh, 2023-02-09:
I designed a voltage divider to downscale the voltage from echo pin (5 to 3 V) and used other available GPIO pins to integrate the sensors.

closed 2023-02-09 by Sharath-Lokesh

imadjamil, 2024-01-02:
> I designed a voltage divider to downscale the voltage from echo pin (5 to 3 V) and used other available GPIO pins to integrate the sensors.

What is the purpose of the voltage divider as the ultrasonic modules work with 5v, no?

Sharath-Lokesh, 2024-01-02:
The voltage from the echo pin of the ultrasonic sensor is 5v. However, the board GPIO pins were operated at 3.3v (5v GPIO pins were already occupied). So, I used a voltage divider to reduce the output from echo pin to 3v.