Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #39: memory access error on server.py

Flomedi, 2022-12-22:
I have the problem that i get a "memory access error" (Speicherzugriffsfehler in german) whenever i try to connect to the server. In windows i get "unpack requires a buffer of 4 bytes" error. I provide screenshots for more clearity.
[image]
[image]

Shaynee-MO, 2022-12-27:
Please install the latest 32-bit Raspberry Pi OS and try again.
https://www.raspberrypi.com/software/operating-systems/

[image]

Blcker12, 2023-01-09:
I have the same issue and it came back after installing the latest greatest image and all updates.
[image]

Blcker12, 2023-01-10:
it is an issue with the picamera2 configuration. When I put the camera part into comments (server.py) the issues is gone. 
Need to check but I think the output configuration is causing it.

fluxens, 2023-08-07:
Did you solve this problem?

Blcker12, 2023-08-08:
Yes. First I got an Raspy Image from the developers which was working. After that I re-installed the orifinal image from www.raspberrypi.com/software/ with the Raspberry imager programm and it work as well... But I still have no clue what was wrong at first trial...

fluxens, 2023-08-08:
@fluxens thanks for the reply. I updated the image and it works for me now as well.

I think this issue can be closed.