Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #40: camera not found error

Sanid98, 2022-12-29:
Just assembled and tested the car. All systems work except for camera does not seem to connect. cables are inserted correctly. here is the error message I get. please let me know if you have run into this and can suggest what to try. It might just be a bad cable

sanid98@raspberrypi:~/freenove_4wd_smart_car_kit_for_raspberry_pi/Code/Server $ python camera.py
[0:35:53.446496540] [1363]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
2022-12-29T13:11:30.320Z | ERROR    | Camera(s) not found (Do not forget to disable legacy camera with raspi-config).
2022-12-29T13:11:30.320Z | ERROR    | Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 123, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 305, in _open_camera
    if self._initialize_camera():
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 271, in _initialize_camera
    raise RuntimeError("Camera(s) not found (Do not forget to disable legacy camera with raspi-config).")
RuntimeError: Camera(s) not found (Do not forget to disable legacy camera with raspi-config).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sanid98/freenove_4wd_smart_car_kit_for_raspberry_pi/Code/Server/camera.py", line 2, in <module>
    picam2 = Picamera2()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 130, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.
sanid98@raspberrypi:~/freenove_4wd_smart_car_kit_for_raspberry_pi/Code/Server $

Shaynee-MO, 2022-12-30:
[Disable_Legacy_Camera.pdf](https://github.com/Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/files/10323640/Disable_Legacy_Camera.pdf)
Please refer to the attachment to disable legacy camera and try again.

developervova, 2024-11-23:
[image]
What should I do if there is no "Legacy Camera" in the menu?