Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #42: Checking if data2==None twice

HarHarLinks, 2023-01-13:
https://github.com/Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/blob/4c513154fb11cae7d11f924c7c11688daf611bd4/Code/Server/server.py#L197

https://github.com/Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/blob/4c513154fb11cae7d11f924c7c11688daf611bd4/Code/Server/server.py#L218

1. data2 is checked twice.
2. data4 is never checked. is this intentional?
3. there are at least these 2 occurrences.
4. please use `is None` instead of `== None` per PEP8 (https://www.flake8rules.com/rules/E711.html)

fire-eggs, 2023-01-13:
Wow! Thank you for this, somehow I never noticed!

fire-eggs, 2024-01-10:
[Finally addressed](https://github.com/Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/commit/271306cc2855256e1575956d72783bda2f066321).

closed 2024-01-11 by HarHarLinks