Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #46: Client crashes with segmentation fault

sjuk, 2023-12-26:
Server is running on Raspberry Pi 4B with latest Raspberry Pi OS (headless).
Client is running on desktop PC with Ubuntu 22.04.

Note that I can reproduce the issue on a notebook with Ubuntu 22.04 which is using Xorg instead of wayland.

Steps to reproduce:
1. Start client.
2. Connect to car server.
3. Open Video
4. Wait for a while enjoying the face detect feature. ;-)
=> Client will crash after a while.

Output:

```
~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi-master/Code/Client$ LANG=C python Main.py
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QFont::setPointSize: Point size <= 0 (-1), must be greater than 0
QFont::setPointSize: Point size <= 0 (-1), must be greater than 0
QFont::setPointSize: Point size <= 0 (-1), must be greater than 0
QFont::setPointSize: Point size <= 0 (-1), must be greater than 0
Server address:192.168.178.35

Connection Successful !
CMD_POWER#7.38

...

QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?

...

CMD_POWER#7.17

QWidget::repaint: Recursive repaint detected
QWidget::paintEngine: Should no longer be called
QPainter::begin: Paint device returned engine == 0, type: 1
Speicherzugriffsfehler (Speicherabzug geschrieben)
```

sjuk, 2023-12-27:
Client
```
~$ getconf LONG_BIT
64

```
Server
```
~ $ getconf LONG_BIT
64
```

sjuk, 2023-12-27:
Client
```
~$ cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 158
model name	: Intel(R) Core(TM) i3-9100 CPU @ 3.60GHz
stepping	: 11
microcode	: 0xf4
cpu MHz		: 800.011
cache size	: 6144 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_ad ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple pml ept_mode_based_exec
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit srbds mmio_stale_data retbleed gds
bogomips	: 7200.00
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 158
model name	: Intel(R) Core(TM) i3-9100 CPU @ 3.60GHz
stepping	: 11
microcode	: 0xf4
cpu MHz		: 3600.000
cache size	: 6144 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 4
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_ad ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple pml ept_mode_based_exec
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit srbds mmio_stale_data retbleed gds
bogomips	: 7200.00
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 158
model name	: Intel(R) Core(TM) i3-9100 CPU @ 3.60GHz
stepping	: 11
microcode	: 0xf4
cpu MHz		: 1257.205
cache size	: 6144 KB
physical id	: 0
siblings	: 4
[... 64 lines cut ...]
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Revision	: b03114
Serial		: 100000005b0c53ec
Model		: Raspberry Pi 4 Model B Rev 1.4
```

sjuk, 2023-12-27:
Client
```
~$ cat /etc/rpi-issue
cat: /etc/rpi-issue: Datei oder Verzeichnis nicht gefunden
```

Server
```
~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2
```

sjuk, 2023-12-28:
Maybe it has something to do with the following error message in the console.
```
QWidget::repaint: Recursive repaint detected
```

Shaynee-MO, 2024-01-04:
Hi! Many thanks for the feedback. We have updated the code. You can download the latest one and try again.

sjuk, 2024-01-07:
Hi! You are welcome. Thank you for taking care. :-)
I downloaded the latest code and tried again. Unfortunately the client is still crashing after a while.

Output:
```
~/Dokumente/code/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Client$ /bin/python /home/sjuk/Dokumente/code/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Client/Main.py
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
Server address:192.168.178.35

Connection Successful !

... CMD_POWER repeated 23 times ...

QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?

... CMD_POWER repeated 78 times ...

Speicherzugriffsfehler (Speicherabzug geschrieben)

```

fire-eggs, 2024-01-10:
It might be helpful if @sjuk could try a couple of tests to narrow things down a little. After performing either test, the questions are: Does the crash still happen? Does the crash take longer to happen?

1. The implication from the original post is that face detection is enabled. A useful test would be to turn off face detection. [I.e. removing linux as a supported platform in Video.py, line 45]

2. Another test would be to change the video update timer. If I understand the [`QTimer`](https://doc.qt.io/qtforpython-5/PySide2/QtCore/QTimer.html) doc correctly, the timer is currently set to fire every 34 milliseconds. I think this may be too frequent an update cycle. I'd try changing line 509 from: `self.timer.start(34)` to e.g. `self.timer.start(100)`. With that change, the video repaint would take place every 100 milliseconds.
 
My hypothesis [purely from code inspection] is the timer is firing too fast. Depending on the system performance, the timer could fire in the middle of a video frame update. Said update will result in a repaint. By firing in the middle of a repaint, the timer could cause the recursive repaint issue.

The "face detection disable" test is to see if the extra work being done to find faces is slowing down the system and/or the video update process. Without said slowdown, the repaint cycle might be fast enough to keep up with the timer.

A timer of 34 milliseconds is presumably attempting to provide video playback at 30 frames-per-second. This might not be achievable on the client, depending on system load, memory speed or other factors. A better locking mechanism around the timer and video update may be required to prevent repaint faults.

fire-eggs, 2024-01-10:
@xmayeur has an [update](https://github.com/xmayeur/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/tree/QT-Repaint-bug-correction) which claims to address a QT repaint bug. It is unclear which QT repaint bug is being addressed, as the code diffs are difficult to parse due to radical reformatting changes.

xmayeur, 2024-01-11:
Sorry for the code reformatting - side effect of an option in Pycharm to reformat layout according to PEP recommendations at file save.

Here how to bring the modifications to correct the issue - line numbers refers to Freenove master branch git, not mine.
The crash is caused by trying to update GUI widgets in another thread that the one where the widgets are defined.
The solution consists in using QT signals to send values from the modifying thread(s) to the GUI one.

    In /Code/Client/Main.py, after the import statements ending line #22

        # Declare new classes to handle the qt Signals

       
        class ProgBar(QObject):
            """
                ProgBar class manages signals for the battery level progress bar, passing integer value       
            """
            sigPB = pyqtSignal(int)

            def send(self, text):
                self.sigPB.emit(text)

        class SigStr(QObject):
            """
              SigStr class manages signals from issuers, passing string values
            """
            sigStr = pyqtSignal(str)

            def send(self, text):
                self.sigStr.emit(text)

    At the end of the __init__() routine of Main.py, ending line #184, declare new signals objects with their event callbacks
    and the callback routine to update the GUI on signal reception

            self.Pb = ProgBar()
            self.Pb.sigPB.connect(self.onPbChanged)

            self.U = SigStr()
            self.U.sigStr.connect(self.onUsonicChanged)

            self.L = SigStr()
            self.L.sigStr.connect(self.onLightChanged)

            def onPbChanged(self, value):
                self.progress_Power.setValue(value)

            def onUsonicChanged(self, value):
                self.Ultrasonic.setText(value)

            def onLightChanged(self, value):
                self.Light.setText(value)

    Modify the recvmassage(self) routine of Main.py, replacing lines #760-768 by:

                    if cmd.CMD_SONIC in Massage:
                        # self.Ultrasonic.setText('Obstruction:%s cm' % Massage[1])
                        self.U.send('Obstruction:%s cm' % Massage[1])
                    elif cmd.CMD_LIGHT in Massage:
                        # self.Light.setText("Left:" + Massage[1] + 'V' + ' ' + "Right:" + Massage[2] + 'V')
                        self.L.send(l"Left:" + Massage[1] + 'V' + ' ' + "Right:" + Massage[2] + 'V'
                    elif cmd.CMD_POWER in Massage:
                        percent_power = int((float(Massage[1]) - 7) / 1.40 * 100)
                        # self.progress_Power.setValue(percent_power)
                        self.Pb.send(percent_power)

That's all, Folks! 😎

Have an excellent day.

Xavier.

sjuk, 2024-01-11:
I performed another test with disabled video and disabled face detect.
I will update the introduction to point out that the issue is not related to video.
```
~/Dokumente/code/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Client$ /bin/python /home/sjuk/Dokumente/code/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Client/Main.py
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
Server address:192.168.178.35

Connection Successful !

... CMD_POWER or CMD_MODE repeated 303 times ...

QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?
Speicherzugriffsfehler (Speicherabzug geschrieben)
```

fire-eggs, 2024-01-12:
@sjuk and @xmayeur : thank you both for the updates! I've really enjoyed working with this project and I think it's really nice to see additional energy and participation.

xmayeur, 2024-01-16:
Code in master branch has been updated - I think we can close this issue.
Enjoyed to work on this too :-)

sjuk, 2024-01-17:
@fire-eggs and @xmayeur : Many thanks for your help. It was a pleasure to collaborate with you. :-)

sjuk, 2024-01-17:
Pull request https://github.com/Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/pull/50 fixed the issue.

closed 2024-01-17 by sjuk