Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #52: servo setup problem

Bahauldien, 2024-01-08:
while assembling the car, when i connect the servos and open the server file on terminal & run the servo.py to set them up to 90 i get this message + error:
Now servos will rotate to 90Â°.
If they have already been at 90Â°, nothing will be observed.
Please keep the program running when installing the servos.
After that, you can press ctrl-C to end the program.
Traceback (most recent call last):
  File "/home/baha/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/servo.py", line 34, in <module>
    pwm=Servo()
        ^^^^^^^
  File "/home/baha/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/servo.py", line 5, in __init__
    self.PwmServo = PCA9685(0x40, debug=True)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/baha/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server/PCA9685.py", line 29, in __init__
    self.bus = smbus.SMBus(1)
               ^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory

im kinda noob with python and linux but im pretty sure i have the right directory open and i connected everything correctly so far

Shaynee-MO, 2024-01-17:
Please run the command to install smbus and try again.
sudo apt-get install python3-smbus

HockeyD8, 2024-01-18:
I have the same problem, and it’s not solved by reinstalling python3-smbus.

fire-eggs, 2024-01-19:
@HockeyD8 , @Shaynee-MO :
Just to confirm: is the I2C interface enabled on your Raspberry Pi? The FileNotFoundError message may occur if the I2C interface is not enabled.

How to enable I2C is discussed in "Step 2 Configuration" on page 37 of the PDF tutorial. Or, 
[here's one tutorial](https://www.raspberrypi-spy.co.uk/2014/11/enabling-the-i2c-interface-on-the-raspberry-pi/) that walks through the process of enabling it via the command line utility.