Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #57: Servos twitch uncontrollably

jimmanuel, 2024-03-03:
I recently assembled my smart car and ran the both `sudo python test.py Servo` and sudo python servo.py` and neither produces the expected results that are described in the manual.  As soon as a signal is sent to the servos they begin rapidly twitching left and right and never reach a stable, stopped position.  They continue twitching even after I execute a `sudo shutdown -P now` to turn off the pi and they don't stop until I turn off the Load Switch on the board.  Any advice of help would be appreciated.

jimmanuel, 2024-03-04:
Sorry, didn’t mean to create the duplicate

closed 2024-03-04 by jimmanuel

Shaynee-MO, 2024-03-07:
Hi! Please run the following commands one by one and send us screenshots of the results.
getconf LONG_BIT
cat /proc/cpuinfo
cat /etc/rpi-issue

If possible, please send a video of the servo twitching to our support email: support@freenove.com.

jimmanuel, 2024-03-07:
here's my command output:

```shell
jim@pi-car:~ $ getconf LONG_BIT
32
jim@pi-car:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d4
Serial          : 00000000b9588dd2
Model           : Raspberry Pi 3 Model B Plus Rev 1.4
jim@pi-car:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 02b371de1a87bbe70cfcbd498527067a7d2ee6ca, stage4
```

I'll send a video shortly, as requested.  I also have some new data - I removed the camera portion from the servo assembly and retested it; for a short while it worked as expected but then started acting up after some time.  When it acts up the servos become incredibly hot to the touch.