Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #58: Servos twitch and don't reach a stable state

jimmanuel, 2024-03-03:
I recently assembled my smart car and ran the both `sudo python test.py Servo` and sudo python servo.py` and neither produces the expected results that are described in the manual.  As soon as a signal is sent to the servos they begin rapidly twitching left and right and never reach a stable, stopped position.  They continue twitching even after I execute a `sudo shutdown -P now` to turn off the pi and they don't stop until I turn off the Load Switch on the board.  Any advice or help would be appreciated.

olivedmt, 2024-07-06:
I have the same problem ?
Any ideas how to fix this ?

jimmanuel, 2024-07-06:
@olivedmt - They sent me some new servos - maybe log a new ticket if your question does not get the support teams attention

closed 2024-07-06 by jimmanuel