Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi #61: Please help, I'm stuck at this point.

andreyccmd, 2024-11-16:
pi@4wd:~ $ cd ~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server-pi5
pi@4wd:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server-pi5 $ sudo python servo.py
Now servos will rotate to 90°.
If they have already been at 90°, nothing will be observed.
Please keep the program running when installing the servos.
After that, you can press ctrl-C to end the program.
Traceback (most recent call last):
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server-pi5/servo.py", line 34, in <module>
    pwm=Servo()
        ^^^^^^^
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server-pi5/servo.py", line 5, in __init__
    self.PwmServo = PCA9685(0x40, debug=True)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server-pi5/PCA9685.py", line 32, in __init__
    self.write(self.__MODE1, 0x00)
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server-pi5/PCA9685.py", line 36, in write
    self.bus.write_byte_data(self.address, reg, value)
TimeoutError: [Errno 110] Connection timed out

andreyccmd, 2024-11-16:
pi@4wd:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@4wd:~ $

Shaynee-MO, 2024-11-19:
Please check the following:
1. Whether the connector is plugged all the way in.
[image]

2. Whether both two switches on the car board are pressed.
    Whether the power and battery indicators are ON.
[image]

3. Whether the wiring of all components is correct.